RoboJackets/roboracing-software #19: Set up distributed ROS launch

sastorer, 2016-09-12:


chsahit, 2017-06-05:
What's the done criteria on this one? There isn't a lot of code to be written for this one, see this link on [distributed ROS](http://wiki.ros.org/ROS/Tutorials/MultipleMachines) and maybe this one to [ssh onto the joule](https://askubuntu.com/questions/305229/whats-the-best-way-to-ssh-to-machines-on-the-local-network). Would getting this to work once be enough to say we have distributed ROS launch set up or is there more?

barulicm, 2017-06-05:
The idea here is to make remote launch as automated as possible. It's not a difficult task to do, but takes lower priority than most of the tasks we have coming up. 
The AutoRally project has a very advanced distributed launch system that is more complex than we need, but shares a lot of the basic functionality as what we're trying to do.
Things we want:
- Minimize manual SSH, this is something roslaunch can handle automatically
- Create machine definitions and setup launch files to launch intelligence nodes on the vehicle
- Create setup script which exports necessary environment variables on dev laptop to define the car as the master machine
- Add roscore as an automatically started program on the joule, so that rosmaster is always available when the car boots up

This is one of those tasks that is simple in principle but more complicated when implementing them on real hardware. But yes, once it is done, it will just be done and easily expandable to our larger platform.

barulicm, 2017-11-14:
@chsahit I don't think this is worth pursuing anymore. Given our research last year into ROS on unreliable networks, there is no good way for us to implement this in such a way that it would actually be useful.

chsahit, 2017-11-15:
Good point, neither IARRC nor AVC have a reliable network that makes this feasible. Closing the issue. Something like this might come up again when we decide to cohost a dedicated gazebo server with IGVC so it was a good learning experience.

closed 2017-11-15 by chsahit