RoboJackets/roboracing-software #35: Implement and test histogram matching for AVC

barulicm, 2016-10-23:
Take a look at this tutorial: http://docs.opencv.org/3.1.0/d8/dc8/tutorial_histogram_comparison.html

Build a node that builds a histogram of the track surface based on a rectangle in front of the car. Then match that histogram against the rest of the image to segment out the track v. non-track regions.

See me for details

barulicm, 2018-01-23:
@chsahit Is this still something we're pursuing?

btdubs, 2018-02-11:
For documentation: current code is loosely based on
http://www.morethantechnical.com/2013/03/05/skin-detection-with-probability-maps-and-elliptical-boundaries-opencv-wcode/
http://www.cse.unsw.edu.au/~icml2002/workshops/MLCV02/MLCV02-Morales.pdf
http://www.sciencedirect.com/science/article/pii/S0031320306002767

I will rework the code to only use a single histogram match (had been using multiple for road detection) to bring it back to being useful for a general color detector. If we need it in the future it will be there.

closed 2018-09-30 by ebretl