RoboJackets/roboracing-software #60: Convert IARRC to use the new Path Planner

barulicm, 2017-02-10:
Now that the path planner is stable and the new IARRC circuit course requires it, we should switch the IARRC code base to use the new planner. This means getting rid of the following nodes in rr_iarrc:
- speed_controller
- steerer_circuit
- steerer_dragrace

Some of the functionality included in these nodes need to be re-thought. The speed controller listens for the stoplight watcher's go signal, and counts the number of times we cross the start/finish lines to automatically stop the car. Additionally, the steering nodes contain different parameters for encouraging the car to drive straight to be more aggressive on the drag race.

evanxq, 2017-02-12:
1. publish planner items under /planner namespace
2. in avc, use topic_tools relay
3. in iarrc, use controller state machine or similar

barulicm, 2017-03-21:
Fixed by #66

closed 2017-03-21 by barulicm