RoboJackets/roboracing-software #65: High-Level Route Selection

barulicm, 2017-03-02:
Implement a high-level planner which picks which route we will take through the course. Should be able to answer questions from the path-planner when multiple paths are detected.

btdubs, 2018-09-30:
As a goal for IARRC 2019, I'd like to say being able to detect the cones at the iarrc intersection and being able to publish something about telling us they are there and to turn into it, otherwise telling us to continue on our normal (straight) best path.

Later this can look like seeing the barrels or ramp at AVC and giving us pre-written instructions to turn into or away from some obstacles.

Daniel-Martin576, 2019-08-25:
Closed due to canceled competition.

closed 2019-08-25 by Daniel-Martin576