RoboJackets/roboracing-software #79: CI tools fail because the compilation tests expect joule mraa to be installed

ebretl, 2017-06-27:


barulicm, 2017-06-27:
I've been poking at this issue. I think @chsahit and I figured out the problem the other day, but we've been a bit busy to implement the fix. I'll be taking care of it either tonight or tomorrow night.

barulicm, 2017-06-28:
Never mind, the line I thought we needed to add is already in the setup script. (See [setup.sh](https://github.com/RoboJackets/roboracing-software/blob/6da768cb263c1513ffaea496d3fca00e7958151e/util/setup.sh#L14)).

I pinged @jgkamat on Saturday about the setup script apparently not running. I'll see what he says.

In the mean time, since the only thing our CI does right now is test if the code builds, it's not adding much while we're all working on the code so frequently. So, I imagine he'll relegate this to a relatively low priority with two competitions (RR & RC) in the next three weeks.

jgkamat, 2017-06-28:
I've just triggered a [complete rebuild here](https://circleci.com/gh/RoboJackets/roboracing-software/140) and the script seems to be installing the dependency correctly. I think that there's something else wrong (besides a missing dependency). Interestingly, the build started to be broken [on a merge commit](https://github.com/RoboJackets/roboracing-software/pull/77) so maybe something went wrong in that merge. I can probably squeeze in some time to look at this tomorrow.

barulicm, 2017-06-28:
@jgkamat That merge commit pulled in the new dependency which is causing the problem. I found the output from the setup script and I'm pretty sure I found the issue. 
I copied the manual install instructions for the mraa library into our setup, but it doesn't include the -y tag on a call to apt-get install, so apt-get aborts after timing out waiting for the user to type yes. Thus, the mraa library is not installed.
I'm going to edit this script and try a rebuild. We'll see if I'm right.

jgkamat, 2017-06-28:
If you make the change and push to a branch, you'll be able to see the results of the build (and follow along) from the branches page: https://github.com/RoboJackets/roboracing-software/branches. 

Just tag me here if you run into anything :)

jgkamat, 2017-06-28:
@barulicm looks like you got a [successful build](https://circleci.com/gh/RoboJackets/roboracing-software/141)!. Closing this issue.

To prevent issues like this in the future, I would recommend adding `set -euo pipefail` to the top of all your bash scripts, which will cause them to exit immediately when there is an error like this one (making this much easier to spot). See: http://redsymbol.net/articles/unofficial-bash-strict-mode/

closed 2017-06-28 by jgkamat