RoboJackets/roboracing-software #80: iarrc color detector does not sense the yellow line well in gazebo

ebretl, 2017-06-27:


ebretl, 2017-06-27:
[image]

ebretl, 2017-06-27:
It is also a bit too zealous with finding white lines, as this pic from the drag race track shows

[image]

barulicm, 2017-06-27:
The color detector is using the same rules as were tuned at last year's competition. 
We can tune it for the simulator if we want to, but the current plan is to tune it for real life again when the car is ready.
Also, the last screenshot you're showing isn't finding any white lines, its flagging lots of noise as blue. There are several reasons for this:
1. The old rules for white were based on RGB. We've switched everything to HSV, but haven't implemented the new white rule yet.
2. The old color detector code ran an erosion procedure to get rid of speckly noise. Each color was processed independently though, so we've left this step out until we can try to find a good middle ground for all colors with real-world testing.

The thing I'm most worried about in the image above, however is the fact that it's completely missing the orange cones.

ebretl, 2017-07-02:
mostly fixed by #82

closed 2017-11-19 by barulicm