RoboJackets/roboracing-software #87: cluster possible paths using something other than DBSCAN

ebretl, 2017-07-26:
DBSCAN is the wrong algo for clustering groups of similar paths. We don't have noise in the data; the density doesn't matter, only the cost does; and it needs to be faster. Not the most important problem right now, but it should be fixed.
1. consider aggregating path results into a discrete grid for O(1) neighbors lookup. (Perhaps also take exactly one path sample per grid space.) FLANN KNN is cool, but maybe it solves a harder problem than we need.
2. consider finding clusters by analyzing the distances between local maxima (instead of grouping valid paths through nearest neighbor searches)

barulicm, 2017-07-27:
I don't remember if we've talked about this before, but would something akin to non-maximal suppression help here?

ebretl, 2017-07-27:
The current version already thresholds by removing paths that are not within a certain proportion of the best one. Does that count as NMS?

barulicm, 2017-07-27:
So in vision algorithms, NMS is used to solve this style of problem where you've got "blurry" answers and are trying to reduce them to single points. So, if our threshold is producing islands of good paths, NMS would be able to give us one control vector for each island.

ebretl, 2017-07-27:
Is this done by repeatedly eroding each island, or by choosing the maximum value from each?

barulicm, 2017-07-27:
The latter

ebretl, 2017-07-27:
Cool. Unfortunately our islands are not the result of gaussian blurs/noise, and they sometimes have multiple peaks per "island". It is usually safe to ignore the lower peaks, but I sometimes run into issues where significantly different path decisions make it into the same island. It is hard to tune the filter threshold to deal with this because the required threshold for the edges of path groups is not very consistent in different environments.

barulicm, 2017-07-27:
What is the dimensionality of our control vectors at this point? Is it still two-stage steering and is velocity factored in at this point in the algorithm?

ebretl, 2017-07-27:
2D right now with velocity factored (perhaps too strongly) into the cost

barulicm, 2017-07-27:
It would be cool to plot the cost function to get a sense of its sparseness and the kinds of features we see in it

ebretl, 2017-07-27:
I have a node for that in the repo

ebretl, 2017-07-27:
It's a Python script

barulicm, 2017-07-27:
Awesome

ebretl, 2017-07-27:
Yeah run planner_plotter.py in rr_common

ebretl, 2017-07-27:
NB I think something has to be uncommented from the planner.cpp so that steering samples and costs are published

ebretl, 2017-08-04:
Just had really good success with a partial rewrite of the planner. I switched the clustering algo to finding local minima of costs, and now it keeps up with the lidar's 15hz quite easily

barulicm, 2017-08-05:
That's awesome! Is this clustering algorithm to the point where we could tell it which branch to choose?

closed 2017-09-04 by ebretl