RoboJackets/roboracing-software #90: Solve naming conflicts against IGVC

barulicm, 2017-08-10:
In trying to move all of my ROS-based code bases to a single catkin workspace, I ran up against two naming conflicts between our code and IGVC's.
Looking through other ROS repositories and the ROS documentation, this conventions and standards surrounding these names are surprisingly inconsistent. Unfortunately, catkin, being an extension of cmake, doesn't know about ROS packages and builds all targets across all packages in the same space.
There are ways to solve this and prevent it long term, which we may want to implement eventually, but for now we can solve these conflicts with the following changes:

1. Rename imu -> mpu9250_imu

2. Delete joystick_driver
   I know this sounds extreme, but we don't actually use this node and haven't for the last two years. The small cars have moved to using the RC radio controller for manual control and the big car will have in car controls which will require their own node anyway.

@chsahit thoughts?

chsahit, 2017-08-10:
The imu change sounds fine. My initial concern about deleting joystick driver was that we used it to watch the car roll around in gazebo. I guess its fine to delete that too however, given that there are better ways to debug if something in simulation was broken.

barulicm, 2017-08-10:
Hmmm... That's a good point. I'd forgotten about the simulator.

JasonGibson274, 2017-08-14:
If you want any changes in the IGVC repo just open an issue on the igvc repo with the convention and we will change to match.

barulicm, 2017-08-15:
The running solution for this problem in some of the distributed repos is to use the package name as a prefix in the target name and rename the output file to remove the prefix. Eg.
```
add_executable(${PROJECT_NAME}/joystick_driver joystick_driver.cpp)
set_target_properties(${PROJECT_NAME}/joystick_driver PROPERTIES OUTPUT_NAME "joystick_driver")
```

This results in a node we can run exactly the same as we do now, but won't conflict with any targets from any other ROS packages.

Note that this solution is not implemented by all ROS packages, and it's not even used consistently within some packages (having some plain and some prefixed targets). We could use prefixes just on the targets that conflict, or implement a new standard of prefixing all targets to eliminate all possible conflicts down the line.

Thoughts? @chsahit @JasonGibson274

JasonGibson274, 2017-08-16:
I would prefer a consistent standard and I think we might want to look at a consistent ros style guide for both igvc and roboracing.

JasonGibson274, 2017-08-18:
@rmkeezer set up a clang format for igvc already we just probably take most of that.

barulicm, 2017-09-05:
The conflicts specifically identified in this issue have been fixed in c2871ea
I have added a comment to #64 to add this new target naming convention to our style guide setup.

closed 2017-09-05 by barulicm

barulicm, 2017-09-05:
**NOTE** The above example is imperfect. Technically the target name generated by that code is not a valid cmake target name.
New suggestion:

```
add_executable(${PROJECT_NAME}_joystick_driver joystick_driver.cpp)
set_target_properties(${PROJECT_NAME}_joystick_driver PROPERTIES OUTPUT_NAME "joystick_driver")
```