RoboJackets/roboracing-software #127: Implement Ultrasonic Sensor Array

btdubs, 2018-02-11:
Firmware for microcontroller control of multiple ultrasonic sensors and a Ros node to output the obstacles detected as a point cloud. The point of this is to use the ultrasonic array to detect obstacles.

The firmware is found in the roboracing-firmware repo under macaroni_ultrasonic_array .
The ROS node is found under ultrasonic_array branch .

btdubs, 2018-02-11:
- [ ] Read CSV from serial

- [ ] Transform points using TF

- [ ] Combine multiple point clouds

- [ ] Publish point cloud

- [ ] Test with arduino

- [ ] Update arduino firmware to flipping sensor check to be slightly faster

- [ ] Update arduino firmware to Atmel chip from board we ordered

ebretl, 2018-09-27:
@btdubs was this issue resolved?

closed 2018-09-30 by btdubs