RoboJackets/roboracing-software #140: Improve Gazebo Track Textures

barulicm, 2018-03-02:
The IGVC team's found some documentation that explains how to actually layer textures in Gazebo models!
http://wiki.ogre3d.org/Materials#Alpha_Splatting

We should try to use this to layer different track markings (ie IARRC lines) on top of a more detailed, seamless road texture. This could make it easier to layout different tracks while retaining realistic textures.

For a reference on the general concept of Alpha Splatting, read this: 
https://en.wikipedia.org/wiki/Texture_splatting

JasonGibson274, 2018-03-03:
This is the method we use for the autorally textures. Let me know if you need help with it when you reach it.

JasonGibson274, 2018-10-23:
1. write a script that takes in an image (RGB) and outputs the correct alpha image (convert anything that is supposed to be road to full transparency. in rr_gazebo/scripts/texture. It will put the image in rr_gazebo/materials 
2. create a gazebo material that looks like this 

material Gazebo/Texture_blend_[track name]
{
  technique
  {
    pass
    {
      ambient 0.1 0.1 0.1 0.1
      diffuse 0.8 0.8 0.8 1.0
      specular 0.01 0.01 0.01 1.0 2.0

      texture_unit
      {
        [use gazebo texture for road here]
        filtering trilinear
        scale .01 .01 
      }
      texture_unit
      {
        texture textures/[you output image]
        colour_op alpha_blend
      }
      texture_unit
      {
        texture textures/[the image of the track that was drawn]
        colour_op_ex blend_current_alpha src_texture src_current
        alpha_op_ex source1 src_manual src_current 1.0
        scale .01 .01 
      }
    }
  }
}

3. create a model that uses that texture

JasonGibson274, 2018-10-23:
Other things I think you should condense all the material files into a single material file.

closed 2019-08-25 by Daniel-Martin576