RoboJackets/roboracing-software #141: Cleanup Planner code

barulicm, 2018-03-06:
Once #138 is done, someone should go through and clean up the planner code. It smells of OOP, but is implemented with free functions and global data. 

Actual planner code and data should be wrapped into a planner class and the main method should be split out into it's own file. While this is being done, special care should be paid towards code cleanliness and style consistency with the rest of the code base.

ebretl, 2018-04-03:
- planner.cpp should contain the publishers and subscribers, as well as an instance of StochasticPlanner
- stochastic_planner.h should contain the definition of the StochaticPlanner class. Method plan() should take in a map of the world as a pcl::PointCloud and output a steering angle. Launch params should be passed into the constructor and held as const members. I *think* that the constructor and plan() are the only members that need to be public.
- stochastic_planner.cpp should contain all the code that implements plan()

JasonGibson274, 2018-04-03:
Since we are restructuring I am going to make a gtest plug here. I would recommend adding test cases for the path planner node. You can look here for how https://github.com/RoboJackets/roboracing-software/tree/eff2613de17a39e8cae093bfe578192ffca4c704/rr_platform/test

ebretl, 2018-04-04:
@JasonGibson274 Great suggestion. One potential issue is that the output of the path planner is not deterministic. With that said, we can do general checks on the outputs. For example, we used to have a problem when the robot was facing a wall; the good paths to the left and right were averaged to go straight into the barrier. It would be a nice regression test to give the planner a map of this situation and check that it decides to turn hard to one side or the other.

barulicm, 2018-04-05:
- gtest can also help with testing individual functions / step of the planner that are more deterministic.
- You can set the random seed in the gtest suite to ensure the same "random" numbers get used every time.

closed 2018-10-30 by ebretl