RoboJackets/roboracing-software #142: Touch screen drivers for Bigoli

chsahit, 2018-03-17:
Link to drivers:
https://drive.google.com/drive/folders/0B88VdpNR67axcWs3emstdGFsck0 

My understanding is that we have to compile them ourselves.

barulicm, 2018-03-19:
We should also save a copy on OwnCloud or some other RJ-managed resource as a mirror.

btdubs, 2018-04-10:
Adding some references for later:
Our monitor: https://www.gechic.com/en-portable-monitor-onlap1303h-overview.html
Walkthrough for compile and install (kinda): https://www.icitouchtech.com/single-post/2015/09/23/How-To-Install-A-Touch-Driver-On-Your-GeChic-Touchscreen-Monitor
In the walkthrough, the member variables referenced have been removed (https://github.com/torvalds/linux/commit/e04a0442d33b8cf183bba38646447b891bb02123#diff-88d50bd989bbdf3bbd2f3c5dcd4edcb9).

ebretl, 2018-05-27:
It turns out that our specific model of the display does not have a touchscreen feature. Leaving this open for when we do get our hands on the touch-enabled version

barulicm, 2018-05-29:
I believe we were given permission over slack to use the screen that Will had initially acquired for the EE room inventory system. It's the same as ours, just the correct touch model.