RoboJackets/roboracing-software #234: Cloning repo in CI is massive

ebretl, 2019-03-14:
Configure Circle to clone with `--depth 1` to reduce unnecessary downloads

jgkamat, 2019-03-14:
Evan Bretl writes:

> Configure Circle to clone with --depth 1 to reduce unnecessary downloads

Note that - if time is what you're trying to optimize, it's possible to cache
the previous checkout, which results in a very fast clone (as the majority
will come from the previous build on a cache server. There's docs for the
method on circleci's website, and robocup's circleci file should have it
configured to some level.

barulicm, 2019-03-15:
I'm pretty sure our CI config does that as well. I copied the `save_src_cache` and `load_src_cache` macros from the RoboCup Circle config.

barulicm, 2019-03-15:
@ebretl Looking through our Circle logs, checking out the code only started taking noticeable time within the last 22 hours.

barulicm, 2019-03-15:
Oooh no. I have found the issue. The cached .git folder is getting put in the wrong place (catkin_ws/.git) so it is re-cloning the whole repo every build

barulicm, 2019-03-15:
I can fix and test this this weekend if you want. If you want to try it yourself, I'm *pretty sure* you just need to fix the `save_src_cache` macro as follows in .circleci/config.yaml

```
defaults: &save_src_cache
  save_cache:
    key: source-v1-{{ .Branch }}-{{ .Revision }}
    paths:
    - "~/catkin_ws/src/roboracing_software/.git"
```

PR #240 (2019-04-09): CircleCI cache fix
This fixes #234 

**Ba-Ba-Ba-Bonus Fix!!** Adds rr_ml to our README documentation. (I needed to change something to test the cache fix).
closed 2019-04-09 by ebretl