RoboJackets/roboracing-software #254: Create "Man vs. Machine" Instructions

Daniel-Martin576, 2019-09-09:
After the _Man vs. Machine_ test, a launch file must be added and basic instructions on how to run the setup created. Basically, this instruction set should be designed so that future members will be able to recreate and improve upon the currently tested perception and drive system.

btdubs, 2019-09-09:
Noting for my self that I'd also like to add better launch configuration for mapping controls to outputs. We will always have steering, accelerator, and braking, so it could be way to quickly select which joystick channel does what.

Future work could include specific buttons like cruise control which would be neat (especially if we incorporated the lidar for automatic stop).

btdubs, 2019-10-26:
Added wiki page for this: https://wiki.robojackets.org/RoboRacing_WiFi_Driving
Still want to add better joystick mapping from launch before closing this.