RoboJackets/roboracing-software #256: Visualize Chassis State in RVIZ

btdubs, 2019-09-10:
**Goal:** A [RVIZ](http://wiki.ros.org/rviz) panel to visualize [chassis_state message](https://github.com/RoboJackets/roboracing-software/blob/master/rr_platform/msg/chassis_state.msg) data including steering, speed, and control board state to allow us to quickly determine how the car control board is doing without reading or echoing the data directly.

**Notes:** The control board reports back steering, speed, and other data, and this is published by a relay node on the topic `/chassis_state`
The speed should be visualized as a car speedometer, the steering visualized as an arrow and/or a steering wheel, and the board state as words with highlight color. The control board state is reported as a number, which represents a state such as `E_STOPPED`, `IDLE`, etc. If the state is `E_STOPPED`, this could be highlighted in red and similar color scheme for other states.
Feel free to experiment with other possibly better ways to visualize this information.

RVIZ panels use the [QT library](https://doc.qt.io/qt-5/) to draw.
There is a readme [HERE](https://github.com/RoboJackets/roboracing-software/blob/master/rr_rviz_plugins/HOWTO.md) that explains how to set up an RVIZ panel, and you can see the [ExamplePanel](https://github.com/RoboJackets/roboracing-software/blob/master/rr_rviz_plugins/src/ExamplePanel.cpp) for a quick start panel (Do not modify ExamplePanel!).

**New Files Created (at least):** `/rr_rviz_plugins/include/rr_rviz_plugins/chassisPanel.h` , `/rr_rviz_plugins/src/chassisPanel.cpp`
**Files Edited (at least):** `/rr_rviz_plugins/plugin_description.xml` , `/rr_rviz_plugins/CMakeLists.txt`

**Returning Member Mentor:** Brian Cochran @btdubs --Ask him questions!
When you believe you have finished the success criteria below, submit a [pull request](https://help.github.com/en/articles/about-pull-requests) to merge your code into master.

**Success Criteria:**
- Panel is available to open in RVIZ
- Displays notification when no `/chassis_state` published
- Speedometer speed is accurate with reported speed
- Steering visualization is accurate with  reported steering angle
- Steering viz and speedometer have numeric associated with the visualization
- State of control board is visualized with color

**Future Work (if you want to give it a try, but don't waste too much time on it):**
- Allow dynamic changing of speedometer range
- Show a small section with average speed and average steering over the past few seconds

oswinso, 2019-09-25:
Link to [chassis_state_message](https://github.com/RoboJackets/roboracing-software/blob/master/rr_msgs/msg/chassis_state.msg) is broken.

btdubs, 2019-09-26:
Good catch. Fixed.

PR #312 (2020-06-25): Feat/visualize chassis state
Fixes #256  Adds a widget to RVIZ called ChassisPanel which shows the heading and speed that /chassis_state is reporting as well as weather it is recording a bag file and the estop is on or off.
closed 2020-06-25 by Daniel-Martin576