RoboJackets/roboracing-software #273: Global Center Line Finder

Daniel-Martin576, 2019-10-07:
Find the centerline of the track after the cold lap. It should be a list of points that produce a closed circuit around the track through the middle of the lane. 

Steps:
- Find the distance of each cell to the closest wall
     - Convert to image and use OpenCV distance transform
     - Try to find something else online
     - Outside the track should be 0 and inside should be some positive x distance
- Use Uniform-cost Search to find loop around the course using the distance as a utility  
     - Set goal point behind you and add a wall between the start point and goal point to prevent the search from going backward
- Smooth the resultant circuit if needed

Daniel-Martin576, 2019-10-07:
Maybe this is useful for the first part: https://github.com/artivis/distance_map

btdubs, 2020-03-23:
General update on this since Daniel wasn't in the meeting today. We can now find the center line through the path, I am just debugging an issue where the final path is not outputting correctly in world coordinates (as compared to coordinates within the costmap).

PR #318 (2020-06-25): Global Center Line
Closes #273 by adding a costmap layer for calculating the center line of the map.

The procedure is to use the free_space as the search space, calculate distance from free cells to obstacles, convert distance to cost by setting highest distance to 0 increasing cost the lower the distance, then uniform cost search.

This did adjust the costmap to be trinary state which adjusted the distance_map cost calculations. I have not yet tested it with out with the other map cost interfaces to make sure they work with trinary state maps.

Down scaling the map speeds search time, at the [... cut]
closed 2020-06-25 by Daniel-Martin576