RoboJackets/roboracing-software #286: Local map coordinate frame should lead the robot

ebretl, 2019-11-13:
We want to use a local costmap for trajectory planning (see #285 ). The center point for this costmap is determined by the pose of at TF frame. We could use the link `base_footprint` for this, which puts the robot at the center of the local costmap. We would like to have the center of this costmap in front of the robot because the planner should not need to know if there are obstacles behind the robot.

Deliverable: ROS node which listens for TF updates and publishes a TF update from `base_footprint` to `local_map_center` at 10 Hz.
- Distance forward from `base_footprint` to `local_map_center` should be specified via rosparam/launch files
- This is not performance-critical so it can by Python instead of C++. Try to keep CPU usage minimal though

Daniel-Martin576, 2019-11-13:
I think this issue might be too small to be a new member project. It is just changing three lines. (40 is just random shift)
   ` <node pkg="tf2_ros" type="static_transform_publisher" name="$(anon tf_static_base_footprint_to_shifted)"
          args="40 0 0 0 0 0 base_footprint shifted_base_footprint']))"/>`
then just make
 `robot_base_frame: shifted_base_footprint`
for local map

ebretl, 2019-11-15:
Oh yeah, that would be a much better way to do it :man_facepalming:. I was picturing a continuously updated world -> shifted_base_footprint

closed 2019-11-15 by ebretl