RoboJackets/roboracing-software #298: Update EVGP Localization

Daniel-Martin576, 2019-12-09:
Since apparently we can't use our GPS's RTK capabilities during comp we need to update our localization by adding lidar scan matching and testing how much our map drifts when we remove the RTK capabilities while testing in Gazebo. 

We might also need to potentially experiment with vision odometry if there still a large error between our predicted pose and the ground truth even with scan matching.

JasonGibson274, 2019-12-10:
Is this adding scan matching to the EKF in robot_localization?

Daniel-Martin576, 2019-12-11:
Yeah, we will be combining all the pose estimations through EKF.

ebretl, 2020-02-16:
I'm planning to work on this because it seems interesting. I can't assign myself to the issue, but I'll post an update once I have some progress

btdubs, 2020-02-16:
Are you planning on adding scan matching or do you have something else in mind?

ebretl, 2020-02-16:
Probably just setting up one of these http://wiki.ros.org/laser_scan_matcher