RoboJackets/roboracing-software #301: Sign Detection

Daniel-Martin576, 2019-12-09:
Last IARRC Comp during the Urban Challenge, it seemed like sometimes the arrowed signs were not being picked up by our current sign detector so in order to improve its reliability we are going to experiment with OpenCV's Feature Matching. There are test day and competition bag files on the [RJ cloud](https://cloud.robojackets.org/apps/files/?dir=/RoboRacing/) of signs with arrows for you to use for testing. We previously tried template matching but testing against distinct 4 templates at multiple sizes was too slow.

btdubs, 2019-12-11:
My recommendation is first try this adjustment to the current code and see how good your results are. Instead of using Hu Moments to compare the contour shapes to known shapes, grab the area of the arrow, scale the known arrow image to that size, and template match.

If that doesn't work, you can try using one of OpenCV's feature matching algorithms between the camera image and the known arrow image. At the very least this will give you the possible image location to template match against, but there are other (possibly better) ways to determine if the image is there. I am also not sure how fast feature matching will run.

btdubs, 2020-03-27:
Messed with feature matching quickly today for fun, and results at best give an area to template match but mostly worse that current. As expected, all the corners on the arrow look exactly alike which make it difficult to match. My recommendation is to simply add template matching over possible matches to verify the match.

Daniel-Martin576, 2020-11-08:
Made a new issue.

closed 2020-11-08 by Daniel-Martin576