RoboJackets/roboracing-software #317: Refactor from custom serial to a Serial ros package

btdubs, 2020-03-11:
To get Rigatoni set up quickly, we copied IGVC's Yost IMU code which required the [Serial package](http://wiki.ros.org/serial). To be consistent, we should move all of our Serial code over to that package instead of our custom Serial code.

Code touched will be all our car's motor relays at least.

oswinso, 2020-03-11:
If you're copying our Yost IMU code, then copy the ones from [this pr](https://github.com/RoboJackets/igvc-software/pull/647) instead of the one on master, because we copy-pasted ours from master and its shit

oswinso, 2020-03-11:
Also, I think since we're both using the Yost IMU and it seems like we're both copy pasting this code, I think this would be a good opportunity to put this in [rj_ros_common](https://github.com/RoboJackets/rj-ros-common) so we don't have to like fix the same issues / add the same features twice.

btdubs, 2020-03-11:
Good point. I'll update the issue to reflect that.
That pr reminds me we also need to implement the diagnostic system to clean up a lot of the terminal spam.